arduino-org/Arduino #1: Zero boot loader source code missing

dpslwk, 2015-03-13:
There is no source code for the bootloader_zero_pro_150226.hex
Any chance this can be added to so other can help use/improve it

quasto, 2015-03-13:
our aim is to release it in a few days

buzztiaan, 2015-04-07:
thats nice

tirithen, 2015-04-07:
Check the original version of the Arduino code base, there are more people there that is able to help you out https://github.com/arduino/Arduino

m131, 2015-04-07:
hi tirithen, on the repository you reported, they don't have the code for the Arduino Zero Pro board...!?
The library to manage that board is present only here: https://github.com/arduino-org/Arduino
If I well understood also the new Bosch shield is present only here and not on the other repository. 
So it seem to be more updated this one...

tirithen, 2015-04-07:
You can read more about the issue on one or all of the following links:
* https://github.com/arduino-org/Arduino/issues/2
* http://hackaday.com/2015/04/06/arduino-ide-forked/

This repo was copied by a company fairly that recently took the name Arduino (that was already taken) and copied all of the code. The original arduino community can be found on one of hte following links:
* http://arduino.cc/
* https://github.com/arduino/Arduino/

m131, 2015-04-07:
thanks for the links I have already read them... but the topic of this issue is regarding Arduino Zero Pro. Can you help me to find the link of the code and library for the new boards in the other Arduino repo? 
I remeber Massimo announced this board on the latest NY Maker faire... Am I right?
It seems the code is not there, I found it only here...
I think this is a repository forked to that one you reported but it is updated with the new boards.

colinta, 2015-04-08:
If `arduino/Arduino` doesn't have the code, you should wait for it to be released there.  Any code here is worthless.

If you bought a Zero board already... yikes.  It may or may not have the correct specs.  It's a rip off, not the real thing.  Check arduino.cc for updates, but the Zero isn't out yet.

m131, 2015-04-08:
ok guys, anyway I started to use it and it works... I'll continue to test and try to use it on my projects... I am using new debug functionality, and the new peripherals ... I'll keep you update about issues on the software... I'll love to open new issues for that... 
BTW is there any date for the code release on arduino.cc website, when the Arduino Zero we'll be available on arduino.cc? Let's start a countdown!!! Give us a date so we can follow it!!!

quasto, 2015-04-10:
@dpslwk we just released the source code of the arduino zero pro bootloader. It will be included in the next release of the Arduino.org IDE, but you can found it here: https://github.com/arduino-org/Arduino/tree/ide-org-1.6.1.x/hardware/arduino/samd/bootloaders/zero/Bootloader_D21
Any comments or improvements are welcome, thank you for your patience.

closed 2015-04-10 by quasto