arduino-org/Arduino #2: Rename this fork and use less confusing versioning

probonopd, 2015-04-05:
This fork of the Arduino IDE should be renamed something else to avoid confusion, and a version number should be chosen that does not suggest that this fork is "ahead of" the original Arduino IDE when in fact it is not. Naming open source projects should follow community best practices.

As for "arduino-org", one can read on https://github.com/arduino-org that "This organization has no public members. You must be a member to see who's a part of this organization". This is quite telling. The real Arduino team is at https://github.com/orgs/arduino/people as everyone can clearly see.

richrad, 2015-04-06:
👍

azide0x37, 2015-04-07:
[image]

ipapadop, 2015-04-07:
What are the reasons for this fork? What is the new functionality that you are offering? What issues is it solving and why isn't it just a pull request?

How about you just grow up and actually contribute something instead of confusing developers?

raster, 2015-04-07:
[image]

twitchyliquid64, 2015-04-07:
Agreed. Its fine that you want to fork it and do what you want, but please dont frustrate and drive away your users with annoying differences in versioning and operation of otherwise practically-identical IDEs.

arjanwiegel, 2015-04-07:
Please do so. Not doing so might degrade the value of the whole Arduino "family"

pwillard, 2015-04-07:
Creating ambiguity will not hide reality. I support this comment.

[image]

pixelk0, 2015-04-07:
[image]

jackmaney, 2015-04-07:
`+Infinity`

arthurcolle, 2015-04-07:
(y)

jakerullman, 2015-04-07:
:thumbsup:

milinda, 2015-04-07:
:thumbsup:

jackmaney, 2015-04-07:
By the way, you (meaning the people who started this phony organization and made this phony fork) know that you've [made the front page of Hacker News](https://news.ycombinator.com/item?id=9333783), right? If you continue, there will be no shortage of people who will constantly point out that you're a bunch of wannabe fakers.

d0ky, 2015-04-07:
:trollface: :+1:

FlyingLotus1983, 2015-04-07:
@jackmaney not just Hacker News, but [Hackaday.com](http://hackaday.com/2015/04/06/arduino-ide-forked/) and a few other places.

ashwinrs, 2015-04-07:
[image]

[image]

[image]

tirithen, 2015-04-07:
@jackmaney posted about this in the largest Swedish Facebook page for developers as well.

cport1, 2015-04-07:
(y)

Mnstri, 2015-04-07:
👍

damcedami, 2015-04-07:
:100:

thoughtpalette, 2015-04-07:
:+1: :man:

jackboot7, 2015-04-07:
[image]

m-o-e, 2015-04-07:
[image] [image]

stefania11, 2015-04-07:
image

fduraibi, 2015-04-07:
```
            _
           /(|
          (  :
         __\  \  _____
       (____)  `|
      (____)|   |
       (____).__|
        (___)__.|_____
```

setola, 2015-04-07:
:+1: come on guys... don't be lame :)

mattlo, 2015-04-07:
[image]

jvt, 2015-04-07:
[image]

listkarta, 2015-04-07:
(y)

clearminds, 2015-04-07:
++

Reiuiji, 2015-04-07:
:+1: :zzz:

edbedbe, 2015-04-07:
I have a joke. How many programmers does it take to agree that a misleadingly named fork should be dealt with?

andreybratukhin, 2015-04-07:
[image]

aemreunal, 2015-04-07:
@edbedbe Every single programmer, apparently.

vivin, 2015-04-07:
[image]

teachop, 2015-04-07:
[image]

hasandz, 2015-04-07:
[image]

djbeadle, 2015-04-07:
👍

RockyTV, 2015-04-07:
You should rename it and just keep the schematics.

alexrj, 2015-04-07:
👍
This is a phishing attempt. Please rename this fork and stop trolling.

webguy16, 2015-04-07:
https://assets-cdn.github.com/images/icons/emoji/unicode/1f44d.png

webguy16, 2015-04-07:
[image]

harrisonpage, 2015-04-07:
mmm delicious dinner mints

derek-virtustream, 2015-04-07:
[image]

sparr, 2015-04-07:
Please stop with the thumb and +1 comments. You aren't helping.

jackmaney, 2015-04-07:
:+1: yet again, just because I can. :smile:

openhardwarecoza, 2015-04-07:
Yes please.  In fact rename your whole scam to something else.  Or just crawl in a hole and die?

mixania, 2015-04-07:
⭐️⭐️⭐️⭐️⭐️

KateAdams, 2015-04-07:
[image] [image]

IrregularShed, 2015-04-07:
I think what Smart Projects can also take from this is that they've lost the hearts and minds of the community on which their future depends.

jackmaney, 2015-04-07:
@sparr: Is it starting to sink in, yet? We'll comment when we please, and there is nothing that you will ever be able to do about it. Understood?

simonmorley, 2015-04-07:
:100: !

GastonRosso, 2015-04-07:
while(1) {
 console.log('+1');
}

GastonRosso, 2015-04-07:
```javascript
while(1) {
  console.log('+1');
}
```

KateAdams, 2015-04-07:
@GastonRosso What the fuck is that shit?

```c
while(true)
{
	printf("+1\n");
}
```

GastonRosso, 2015-04-07:
@KateAdams Just for you:

```
10 PRINT "+1"
20 GOTO 10
```

KateAdams, 2015-04-07:
@GastonRosso Better than that Java(script) garbage.

jackmaney, 2015-04-07:
Hmmmm...not enterprisey enough. Could you include some XML? Maybe a few factories?

GastonRosso, 2015-04-07:
@jackmaney Of course, we will need two analysts and five sr devs.

curtisz, 2015-04-07:
[disparaging comment]

[181 more replies]