arduino-org/Arduino #3: Arduino Zero Pro alignment problem

jnorre, 2015-04-07:
When accessing less than 32bit variables in classes, the Zero Pro hard faults.

This very small example works on Arduino Due, but fails on Arduino Zero Pro:
~~~
// Uncomment line below for Arduino Due programming port
// #define Serial5 Serial

class failingClass {
  public:
    bool x; // Change type to e.g. 'int' and it works on both Zero Pro and Due
    failingClass() {
      x = false;
    };
};

failingClass* myTest;

void setup() {
  Serial5.begin(9600);
  Serial5.println("Before init");
  myTest = new failingClass();
  Serial5.println("After init"); // We never get here, when the type in the class is 'bool'
}

void loop() {
  // put your main code here, to run repeatedly:
}
~~~
While Cortex-M0 does not allow unaligned access, it was my understanding that the compiler would work around this, either by making code to access the data or to enlarge the data to fit the alignment limitations (i.e. making the bool 32 bit in size).

tirithen, 2015-04-07:
This is not the original Arduino code base but a copy made by a company that took the same name you will probably get more help at https://github.com/arduino/Arduino

mbanzi, 2015-04-08:
Just FYI the code for the Arduino Zero is an unauthorised release of a very old version of the SAMD21 work done by Arduino (the real one) and ATMEL. Unfortunately the code is full of bugs. We will be releasing our official code in a few days when the real Arduino Zero comes out. I'm sorry you guys are having problems because of this code but we couldn't stop them.

m131, 2015-04-08:
Massimo give us a date so we can start a countdown!!! 
You announced this board last September so we are looking forward to see when it comes out.

jnorre, 2015-04-08:
> tirithen: This is not the original ...
I know that, but I wanted to direct the problem at those who sold me the Zero Pro boards ;-)

> mbanzi:
Thanks for the update. I look forward to seeing the code from "the real one" and to get some "non-pro" zero boards :-)  But, like m131, I am curious about the time frame...

eleceng82, 2015-04-17:
To Jnorre:
Boolean or bool type has size of byte then convert boolean to byte. When you assign x variable replace this: x=false; with this: byte(x)=false; and code works.

PaulStoffregen, 2015-04-21:
FWIW, I just tried this code on a Teensy-LC, which also uses ARM Cortex-M0+.  After changing Serial5 to Serial, and adding "while (!Serial); " it works fine.  The compiler does align things properly.

alfran, 2015-05-21:
To fix this Issue you have to edit row number 70 in ..\hardware\arduino\samd\Platform.txt

like this: 

recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -save-temps -mcpu={build.mcu} "-T{build.variant.path}/{build.ldscript}" "-Wl,-Map,{build.path}/{build.project_name}.map" -o "{build.path}/{build.project_name}.elf" --specs=nano.specs "-L{build.path}" -Wl,--start-group -lm -lgcc -Wl,--end-group -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group "{build.path}/syscalls.c.o" {object_files} "{build.path}/{archive_file}" -Wl,--end-group -Wl,--section-start=.text={build.section.start}

This fix will be inserted in the next release of the IDE.

closed 2015-05-21 by alfran