arduino-org/Arduino #4: Character modifying key combinations do not work

stolendata, 2015-04-07:
Character modifying key combinations such as tilde (alt-gr + ~), umlaut (¨) and other accents do not work in the IDE. The modifying keystroke is accepted, but the following keystroke never outputs a character; it's impossible to write the ~ operator.

The problem is in both 1.5.8 and 1.6.1, and I have only tested this on Linux Mint 16 and 17.1 32-bit.

tirithen, 2015-04-07:
I would suggest that you have a look at the original version of the source code at https://github.com/arduino/Arduino and post issues there as there is many times more developers using the original version of the source code.

quasto, 2015-04-13:
@stolendata thank you for reporting this issue. We managed to reproduce this problem and we will fix it as soon as possible.

quasto, 2015-04-15:
@stolendata we fix this issue and we'll deploy it in the next IDE release. 
Thanks again!

stolendata, 2015-04-15:
It's all of us who thank you :)

closed 2015-04-17 by quasto

WestfW, 2015-04-23:
Was there an error committing this change?  The file history shows only a comment change...

quasto, 2015-04-23:
Please see [there](https://github.com/arduino-org/Arduino/commit/5d429fa62f7e2907bfc568a3abc06a2ffad14a12) the previous commit

WestfW, 2015-04-23:
Ah hah!  There it is.  Thanks.  (I did look for prior patches, but I guess I'm still getting used to github (vs Google Code), and couldn't find it.) ( The "nnnn commits" heading on https://github.com/arduino-org/Arduino is a clickable link.  (Duh.  Sigh.))