arduino-org/Arduino #7: Space character in path name for sketch

sgk, 2015-04-09:
Somebody claims that sketch download for Zero Pro fails if the sketch path name has space character.

This user is using Mac OSX 10.10. The sketch folder is under `/Applications/Arduino Folder/`.
```
/Applications/Arduino Folder/Arduino1.7.0.app/Contents/Java/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: warning: section `.bss' type changed to PROGBITS

usage: dirname path
/Applications/Arduino Folder/Arduino1.7.0.app/Contents/Java/hardware/tools/OpenOCD-0.9.0-dev-arduino/bin/openocd: line 3: /Applications/Arduino: No such file or directory
```

http://blog.livedoor.jp/hardyboy/archives/7872448.html

jackmaney, 2015-04-09:
[Keep in mind that this isn't the real repo for the Arduino IDE](https://github.com/arduino-org/Arduino/issues/2).

quasto, 2015-04-10:
@sgk thank you for report this issue. seems that it's related to openocd, only when you are downloading the sketch through the programming port, in the conditions you mentioned. If you are in the same conditions we suggest for the moment to use a path without space characters. we are investigating and hope to fix it in the next release of the Arduino.org IDE.

tirithen, 2015-05-07:
@jackmaney :+1:

sergiotomasello, 2016-03-01:
@sgk the fix will be included in the next release of the IDE. Thank you for reporting it.

closed 2016-03-02 by sergiotomasello