arduino-org/Arduino #10: Please stop doing this !

incyi, 2015-04-21:
Why do we have now https://github.com/arduino and https://github.com/arduino-org ? Nobody likes this ! You are killing together the Arduino community ! All the new comers will stop being interested in Arduino.

Isaac96, 2015-04-23:
:hamburger: 
:smiley:

bobricius, 2015-04-24:
agree

probonopd, 2015-04-29:
See #2

Isaac96, 2015-04-29:
@incyi 
I agree. 
:law

Testato, 2015-05-05:
(y)

astrobeed, 2015-05-05:
@incyi 
I Agree.

gpb01, 2015-05-05:
@incyi 
Totally agree !!!

Bikky-kun, 2016-07-01:
"Why do we have now https://github.com/arduino and https://github.com/arduino-org ?"
-
This might have something to do with the "Warning" that CC's Arduino IDE shouts out, when connecting a board manufactured by ORG. Such behavior is usually a precursor to more mean doing.
-
So I do understand as to why the IDE was forked.
-
Although the numbering really sucks and confuses more than it helps. Then again, any numbering would be confusing.

Bikky-kun, 2016-07-01:
"Why do we have now https://github.com/arduino and https://github.com/arduino-org ?"
This might have something to do with the "Warning" in CC's IDE on ORG Manufactured Boards. This is just mean behavior and usually a precursor for more.
But the numbering really confuses and therefor sucks. Then again, any numbering would confuse in this case.

Isaac96, 2016-07-05:
I think the warning was removed?
CC is at version 1.6.9 now.

closed 2018-02-16 by incyi