arduino-org/Arduino #12: [Ethernet 2 Library] DHCP Response Latency of IP Address

elsa0919, 2015-06-03:
I got an Arduino Ethernet Shield 2 from this distributor: 
https://www.switch-science.com/catalog/2270/

Follow by the instructions I downloaded the Arduino IDE 1.7.2 and have tried the build-in example DhcpAddressPrinter with Ethernet 2 Library.

It works, it does have return the ip address of my Ethernet 2 shield.

But somehow it is not a very quick response to get this ip address...
I almost wait for 30 seconds then get the response.
It seems quite long to get the response...
Do you have any idea of this phenomena?

*My Testing Environment (the link has specified where I got it)*
- [Arduino UNO SMD Rev 3 (A000073)](http://akizukidenshi.com/catalog/g/gM-07383/)
- [Arduino Ethernet Shield 2 (A000024)](https://www.switch-science.com/catalog/2270/)
- [Arduino IDE 1.7.2](http://www.arduino.org/downloads/previous-releases)

quasto, 2015-06-22:
Hi @elsa0919 
in the example that you are running try to comment this code in the *setup* function
```
 while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }
```

elsa0919, 2015-06-23:
Hi quasto,

Thanks for your advise.
I've tried to comment out that part. It's not working.
A very long delay (30~40seconds) added at the beginning of setup did help get a quick DHCP response.
But about 40 seconds to set up is still taking too long.

quasto, 2015-06-23:
@elsa0919 we could not reproduce the problem, in our test environment the delay is of few seconds (3~5). do you have the opportunity to try it in another environment? for example with a different router.
thank you

elsa0919, 2015-06-23:
Hi quasto,

Yeah it may be the problem of the router. I've tried another router, the Arduino did get a quicker DHCP response around 15~20 seconds at the very first time to boot up right after uploading the sketch.
Pulling out the USB cable (power source) then plugging it in without uploading the sketch again will get a fast DHCP response within 3~5 seconds.

quasto, 2015-06-29:
hi @elsa0919, 
happy to hear that the issue is solved. maybe the first time the router (15~20 seconds) had to "record" the new device with its mac address. the second time (3~5 seconds) the router already "knew" the device and the response was quicker.
regards
quasto

closed 2015-06-29 by quasto