arduino-org/Arduino #14: Arduino IDE v1.7.6 does not start

csystem-it, 2015-07-10:
I have just installed Arduino IDE v1.7.6 on my PC (OS Win7 Ultimate SP1 64bit).
The program does not start correctly: it shows the splash screen and then stops. 
There is no javaw process in the task manager. 

I have the same problem on another PC with Win8 64bit.

I have launched arduino_debug.exe from the command prompt.
It does not start and shows with the following error:
"Error occurred during initialization of VM
Could not reserve enough space for 1048576KB object heap"

In order to start the program correctly, I have to modify the content of files "arduino.l4j.ini" and "arduino_debug.l4j.ini".

Original content:
-Xms128M
-Xmx1024M

Modified content:
-Xms128M
-Xmx512M

Could you check this issue?

Thanks.

CSYSTEM-IT

artynet, 2015-07-10:
The Arduino IDE needs a **x86** Java Runtime Environment installed on your system to properly work. Please download the Java 7 JRE update 80 (7u80) for x86 systems, if not installed on your machine, from the ORACLE website :

http://www.oracle.com/technetwork/java/javase/downloads/jre7-downloads-1880261.html

Let us know if that worked for you, and of course get in touch with us for further hardware/software assistance.

    Kind Regards

csystem-it, 2015-07-13:
We have already installed Java 8 Update 45 on both PCs.

If I set the following parameters in "arduino.l4j.ini" file:
-Xms128M
-Xmx512M
then the program starts correctly.

The problem seems to be the -Xmx1024M parameter.

artynet, 2015-07-13:
That might depend on you system setup (i.e. amount of available ram). We always recommend our users to run the IDE with **java7u80** for **x86** platforms running on their system and after doing this they usually solve the startup issue.

csystem-it, 2015-07-13:
We have 12 GB of RAM on both PC and the OS is 64bit.
We have migrated from java 7 to java 8 as recommended by Oracle.
Maybe the problem is related to the new version of java. 
We will try with the older version.

Thanks.

susan-parker, 2015-08-06:
Awesome, YES!!!

I have had this issue for some months... 

Changing the default values to:
-Xms128M
-Xmx512M

Then it works!!!

Being a hardware person, I have next to no experience with the whole Java thing. 

Of course, there are other issues... but at least the IDE now runs.

Many thanks.
Best,
Susan.

closed 2016-03-02 by sergiotomasello