arduino-org/Arduino #50: Include JavaAppletPlugin.plugin  on Arduino 1.7.10 for macOS 10.11.5

rei-vilo, 2016-06-20:
Please include `JavaAppletPlugin.plugin` as as in other Arduino releases, so Arduino 1.7.10 will be self-contained for macOS 10.11.5.

Apple still supplies Java 6. Java 7 is no longer available, only Java 8 at Oracle.

See [Arduino 1.7.10 on macOS 10.11.5](http://www.arduino.org/forums/arduino-ide/arduino-1-7-10-on-macos-10-11-5-585).

Thank you!

rei-vilo, 2016-06-20:
Same issue after having installed [Java SE Development Kit 7 Downloads](http://www.oracle.com/technetwork/java/javase/downloads/jdk7-downloads-1880260.html).

artynet, 2016-06-21:
Please download the latest update of the jdk7 series from our download servers :

http://download.arduino.org/tools/jdk-7 ... sx-x64.dmg

I always test our OSX releases on a machine running osx El Capitan and it never occurred to me such an issue. No need to set the JAVA_HOME environment variable since it is set by default at the end of the installation.

Please first also download and install this :

https://support.apple.com/kb/DL1572?vie ... cale=en_US

providing the missing java6 environment for OSX and still needed to run some programs.

Arturo

rei-vilo, 2016-06-21:
No luck. 

After reinstalling both Java 7 and Java 6 as suggested, Arduino starts but doesn't display any window for the editor.

[image]

artynet, 2016-06-21:
try removing all Arduino settings with this utility :

https://freemacsoft.net/appcleaner/

and then re-launch the application. Try also to empty the java system cache....as previously told, we have never experienced such an issue on machines running "El Capitan" as other users as well...

sergiotomasello, 2016-06-21:
@rei-vilo try to run `Arduino.app/Contents/MacOS/Arduino` for launching the IDE in debug mode and let us know.
Regards

rei-vilo, 2016-06-21:
Removing the Arduino settings and system entries with AppCleaner did the trick. I suspect macOS built-in security management. 

However, I reiterate my suggestion to include `JavaAppletPlugin.plugin` as in other Arduino releases, so Arduino 1.7.10 will be self-contained for macOS 10.11.5.

sergiotomasello, 2016-06-23:
Thank you @rei-vilo for your suggestion.
Regards

closed 2016-06-23 by sergiotomasello