arduino-org/Arduino #52: EEPROM.length function removed (mising) from IDEs 1.7.0 to 1.7.10, Why?

DouglasBair, 2016-07-17:
I have developed sketches in Arduino.CC's IDEs from version 1.66 to 1.69 with the useful "EEPROM.length()".  When I 'upgraded' it doesn't exist!  
Is there no regression testing, no concern for continuity, or backward compatibility?
The comments/responses to this on the forum and blogs show no intent to reinstate it.
Please fix this if you want us to have any trust in your product.
I will stick with the original Arduino group's software until I see some progress!
Ref: www.arduino.org/forums/arduino-ide/eeprom-library-456

sergiotomasello, 2016-08-29:
@DouglasBair #61 this PR will fix your issue

DouglasBair, 2016-09-26:
I have tested IDE V1.7.11 relavant to the issue, it compiled, and worked in my sketch that relies on
'eeprom.length'.
Thanks, Doug

closed 2016-09-26 by DouglasBair

sergiotomasello, 2016-09-26:
Thank you for report this issue.
Regards