arduino-org/Arduino #69: Make IDE installable alongside with Arduino IDE from arduino.cc on Windows (and other OS's?)

theoneone, 2016-09-23:
The newer boards like the M0 are not part of the IDE from .cc, on the other hand, many new boards are available only there. So, it would be convenient to have both IDEs side by side.

Now the installer argues, there is already an IDE present, so it is to uninstall and/or to update. This happens vice versa.

Please split up the identification from the Arduino IDE from arduino.cc, since this is a fork and no successor.

DouglasBair, 2016-09-23:
I also regret the need for a split.  '.Org' violated '.cc' by labeling their
version of the IDE to be
(seemingly) the next to follow any 1.6.xx.
 
I a please to say I am testing both the original company's IDE 1.6.11 (with
nice improvements)
and '.Org's' 1.7.11 that has both new fixes (and new bugs.
 
The '.org' would like for you to surmise that they and their IDE 17.xx are
the succesor when they are not.
 
My opinion is that their products, documentation, testing and support are
not as good as the original 
company called Arduino.cc!
 
Douglas Bair

  _____  

IDE from arduino.cc on Windows (and other OS's?) (#69)

<https://github.com/arduino-org/Arduino/issues/69> it on GitHub, or mute
jEvJ01Dks5qs-qugaJpZM4KFE_9> the thread.
<https://github.com/notifications/beacon/ARmIyF0yV0B4LcUjAukQqDq7fqi15f9Yks5
qs-qugaJpZM4KFE_9.gif> 

  _____  

No virus found in this message.
Checked by AVG - www.avg.com
Version: 2016.0.7797 / Virus Database: 4656/13068 - Release Date: 09/23/16

WilliamFeely, 2016-12-21:
Actually, now due to the settlement, the two companies will merge, so the best course of action now is to merge the two codebases.