arduino-org/Arduino #70: Arduino M0 stalls if using delay(ms) or delayMicroseconds(mics) in an initializer

theoneone, 2016-09-23:
If the initializer, placed before the setup() function, contains one of delay(ms) or delayMicroseconds(mics), the controller freezes immediately after power on. In this state the USB is not accassible, so it's difficult to resume from this state.

here is an example:
`#include <LiquidCrystal.h>
// this is a library, which calls delayMicroseconds in its constructor

LiquidCrystal lcd(12, 11, 5, 4, 3, 2); // here is delay... called in a sub-function

void setup() {
  pinMode(13, OUTPUT); // this code id from the blink example, just to indicate if the code is running

  // set up the LCD's number of columns and rows:
  // this is the method call containing the delay... - it has been called in the constructor with some
  // default values. Edit the constructor in the libraray:
  //  removing the line "begin(...);" will bring the board back to life.
  lcd.begin(16, 1);

  lcd.print("hello"); // show a message on the LCD
}

void loop() {
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000); // wait for a second
  digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
  delay(1000); // wait for a second
`
The result is no blinking LED, no message on the LCD and no serial connection, until the library is modified as decribed.

Finally the board freezes if the constructor is of this shape:
`MyClass::MyClass() { // e.g. "LiquidCrystal(x,x,x,x,x)" in the example above 
  // none or some statements
  delay(1); // same for "delayMicroseconds(...);"
  // occasionally some other statements
}
`

alfran, 2016-11-15:
Hi theoneone,
The LiquidCrystal library is not compatible with the M0/M0pro/Tian boards. You have to use the LCD Library.

Ciao.

Francesco

theoneone, 2017-04-09:
Hi Francesco,

you shouldn't close this issue so fast.

The LiquidCrystal library was just an example in which place the call of the delay function fails. LiquidCrystal works like a charm, since I removed the delay function from the initializer.

The M0 freezes always, if you do a call to delay or delayMicroseconds in an initializer, like in the example above. Its definitely no bug related to the LiquidCrystal library.

alfran, 2017-04-10:
I apologize for the misunderstanding, I thought that the problem was related only to LiquidCrystal. I will fix it asap

Thank you 

Francescco