arduino-org/Arduino #79: Compile errors in templates since 1.8.0

alexey-vladilenov-sokolov, 2017-01-21:
Hallo all,
it looks like C++ compiler was changed in Arduino IDE (of arduino.org)  since 1.8.0.

I start getting errors in templates with non-type parameters.

```
template<char* Ptr> void print1() { Serial.println(Ptr);} // that's GOOD, both compiles and runs
char a[] = "Hi1";

class Runnable { public: void run(){Serial.println("Hi2");} };
template<Runnable& ref> void print2() { ref.run();}   // compile error since 1.8.0: 'Runnable' has not been declared
Runnable r;
void setup() {  Serial.begin(9600);}
void loop() {  print1<a>();  print2<r>(); }
```

lollo03, 2017-01-30:
i have the same error

andrea-83, 2017-01-31:
Hi all,
move class declaration at the top and then it will compile correctly.
`
class Runnable { public: void run(){Serial.println("Hi2");} };
template<char* Ptr> void print1() { Serial.println(Ptr);} // that's GOOD, both compiles and runs
char a[] = "Hi1";
template<Runnable& ref> void print2() { ref.run();}   // compile error since 1.8.0: 'Runnable' has not been declared
Runnable r;
void setup() {  Serial.begin(9600);}
void loop() {  print1<a>();  print2<r>(); }`

alexey-vladilenov-sokolov, 2017-01-31:
_andrea-83, thanks, it's a good guess!_ 

Now I can propose a **more funny solution**: just move the first (totally unrelated) template statement below, below that template of question... Compiles.

Truely, I don't like these hacks. Curiously, the situation reminds me of preprocessor and C-macros.
I hope nobody could parse templates with preprocessor  :-)

andrea-83, 2017-01-31:
I know the "funny solution" isn't the real solution to your issue :-)…it was just a way to get you going forward