arduino-org/Arduino #80: CiaoRest Client upload to ThingSpeak problem

christoschronopoulos, 2017-03-10:
Using [this example](Arduino/libraries/Ciao/examples/CiaoRestClient-ThingSpeak/CiaoRestClient-ThingSpeak.ino) I can upload two fields to my ThingSpeak channel. However, it is not possible to add 2 more fields and upload data for all of them. What am I missing? The code works for any combination of 2 fields.

`String uri = "/update?api_key=";
    uri += APIKEY_THINGSPEAK;
    uri += "&field1=";
    uri += String(weight, 3); // 3 decimals to thingSpeak
    uri += "&field2=";
    uri += String(pump);
    uri += "&field3=";
    uri += String(temperature,1);
    uri += "&field4=";
    uri += String(humidity);

    Ciao.println(F("Send data on ThingSpeak Channel"));

    CiaoData data = Ciao.write(CONNECTOR, SERVER_ADDR, uri);
    delay(100);

    if (!data.isEmpty()) {
      Ciao.println( "State: " + String (data.get(1)) );
      Ciao.println( "Response: " + String (data.get(2)) );
    }
    else {
      Ciao.println("Write Error");
    }
  }`

bhagman, 2017-03-15:
I've tested this with multiple fields, and it all works fine for me.  I'm thinking that you missed enabling the extra fields in the ThingSpeak interface.

[image]

You have to make sure that the other fields are checked for them to show up.  Can you confirm you have them enabled?

christoschronopoulos, 2017-03-15:
I had enabled the fields but I think that the real problem is Uno memory issues. When I include the UnoWiFiDevEd library, the dynamic memory reaches its limits and the program does not function as it should.

bhagman, 2017-03-15:
Ok - I won't be able to test further without seeing more of your sketch.  You must have more going on that is using up the RAM.

christoschronopoulos, 2017-03-15:
When I create a new sketch, just adding the Uno WiFi library leads to a 50% Dynamic Memory usage.