arduino-org/Arduino #81: RTCInt.cpp needs tweak

Piruthivi, 2020-01-01:
In RTCInt.cpp day setting function and month setting function doesn't support for day '1' and month '1'.
The condition needs to be changed from 
_if((day > 1) && (day <=31))_    to    if((day >= 1) && (day <=31))
and
_if((month > 1) && (month <=12))_   to   if((month >= 1) && (month <=12))

per1234, 2020-01-01:
This repository doesn't contain a file of that name. I see there is an RTCInt.cpp in the arduino-org/brackets-arduino repository:
https://github.com/arduino-org/brackets-arduino/blob/master/libraries/RTC/src/RTCInt.cpp
which does have the bug you mention.

That is apparently part of an Arduino extension for the Adobe Brackets editor. Is that the one you're using?

Piruthivi, 2020-01-02:
thanks for information.

closed 2020-01-02 by Piruthivi